correll/advancedrobotics #10: Provide wireless networking to Jetson

correll, 2015-08-27:


rmaheshkumarblr, 2015-09-18:
Using link: https://nims11.wordpress.com/2012/04/27/hostapd-the-linux-way-to-create-virtual-wifi-access-point/ for reference.

sudo  apt-get update && sudo apt-get install hostapd

ubuntu@tegra-ubuntu:~$ cat /etc/hostapd/hostapd.conf 
#sets the wifi interface to use, it is wlan1 in our case ( D.Link AP )
interface=wlan1
#driver to use, nl80211 works in most cases
driver=nl80211
#sets the ssid of the virtual wifi access point
ssid=jaguar_ap
#sets the mode of wifi, depends upon the devices you will be using. It can be a,b,g,n. Setting to g ensures backward compatiblity.
hw_mode=g
#sets the channel for your wifi
channel=6
#macaddr_acl sets options for mac address filtering. 0 means "accept unless in deny list"
macaddr_acl=0
#setting ignore_broadcast_ssid to 1 will disable the broadcasting of ssid
ignore_broadcast_ssid=0
#Sets authentication algorithm
#1 - only open system authentication
#2 - both open system authentication and shared key authentication
auth_algs=1
#####Sets WPA and WPA2 authentication#####
#wpa option sets which wpa implementation to use
#1 - wpa only
#2 - wpa2 only
#3 - both
wpa=2
#sets wpa passphrase required by the clients to authenticate themselves on the network
wpa_passphrase=changeme
#sets wpa key management
wpa_key_mgmt=WPA-PSK
#sets encryption used by WPA
wpa_pairwise=TKIP
#sets encryption used by WPA2
rsn_pairwise=CCMP
ubuntu@tegra-ubuntu:~$

Jorge-C, 2015-09-21:
Please enclose code and configuration files between triple backticks to make it easier to read, like this:

    ```
    $ echo
    ```

Thanks!

rmaheshkumarblr, 2015-09-24:
The Wireless USB from D-link that we have DWA-130 does not support the nl80211 driver.
Hence I am not able to create access point using it. Also the adapter is not supported by the grinch kernal. Tried hostapd with various options and it does not work.

I could get the Access point working using hostapd  with Intel Wireless PCI card

Would like to have a USB wireless module purchased. Selected one that has external removable antennas.

www.amazon.com/Rosewill-Wireless-Adapter-External-RNX-N250UBE/dp/B006WWN016/

(Rosewill RNX-N250UBE - Wireless N300 Wi-Fi Adapter - IEEE 802.11b / 11g / 11n, (2T2R), Up to 300 Mbps Data Rates, USB 2.0 Interface, Two 5 dBi High Power Detachable Antenna )

It uses rtl8192cu driver which is supported by the Jetson.

correll, 2015-09-24:
Why are we not just sticking to the mini PCIe to create an access point?

rmaheshkumarblr, 2015-09-24:
[image]

The above digram shows the implementation that Steve suggested.

We are planning to create a bridge over the ethernet(eth0) and wireless(wlan0 - Wireless through USB).
This way the access point can be used to connect to the Tegra even for working on it remotely. Multiple people can access the tegra over SSH connectivity. Also it can help us to navigate the robot through the access point rather than the internet when there is no internet connectivity. ( We can have the option to navigate the robot over the internet when there is internet connectivity)

rmaheshkumarblr, 2015-10-29:
Currently using the PCIe Wireless card to create the access point.

```
ubuntu@tegra-ubuntu:~$ cat /etc/network/interfaces
auto lo br0

iface lo inet loopback

allow-hotplug eth0
iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual

iface br0 inet static
	bridge_ports eth0 wlan0
	bridge_waitport 0
	address 192.168.205.1
	network 192.168.205.0
	netmask 255.255.255.0
	bridge_fd 15
	bridge_stp on

ubuntu@tegra-ubuntu:~$
```

```
ubuntu@tegra-ubuntu:~$ cat /etc/dnsmasq.conf 
interface=br0
# Specify starting_range,end_range,lease_time
dhcp-range=192.168.205.100,192.168.205.254,72h
# dns addresses to send to the clients
#server=8.8.8.8
#server=8.8.4.4
ubuntu@tegra-ubuntu:~$ 
```

```
ubuntu@tegra-ubuntu:~$ cat /etc/hostapd/hostapd.conf
#sets the wifi interface to use, is wlan0 in most cases
interface=wlan0
bridge=br0
#driver to use, nl80211 works in most cases
driver=nl80211
#sets the ssid of the virtual wifi access point
ssid=jaguar_ap
#sets the mode of wifi, depends upon the devices you will be using. It can be a,b,g,n. Setting to g ensures backward compatiblity.
hw_mode=g
#sets the channel for your wifi
channel=6
#macaddr_acl sets options for mac address filtering. 0 means "accept unless in deny list"
macaddr_acl=0
#setting ignore_broadcast_ssid to 1 will disable the broadcasting of ssid
ignore_broadcast_ssid=0
#Sets authentication algorithm
#1 - only open system authentication
#2 - both open system authentication and shared key authentication
auth_algs=1
#####Sets WPA and WPA2 authentication#####
#wpa option sets which wpa implementation to use
#1 - wpa only
#2 - wpa2 only
#3 - both
wpa=3
#sets wpa passphrase required by the clients to authenticate themselves on the network
wpa_passphrase=changeme
#sets wpa key management
wpa_key_mgmt=WPA-PSK
#sets encryption used by WPA
wpa_pairwise=TKIP
#sets encryption used by WPA2
rsn_pairwise=CCMP
ubuntu@tegra-ubuntu:~$ 
```

rmaheshkumarblr, 2015-11-15:
Need to buy this Wireless Adapter:

www.amazon.com/Rosewill-Wireless-Adapter-External-RNX-N250UBE/dp/B006WWN016/

rmaheshkumarblr, 2015-11-23:
We got the Wireless Router. Got it working with the Grinch Kernal.
It would appear as WLAN10