correll/advancedrobotics #13: Install drivers for OccamVision cameras

correll, 2015-08-27:
Drivers for Ubuntu 14.04 / ROS Indigo are available from OccamRobotic's website

mira67, 2015-08-31:
Downloaded "ros_indigosdk-2.0.0.tar" and followed the instructions from readme.txt
Tried install the package on Ubutun 14.04 with ROS, got the following error from cmake error log:
/usr/bin/ld: cannot find -lpthread
Anyone has come across same issue?
(Could be the configuration of my ubutun, will test the installation on Jetson.)

RoRa9362, 2015-08-31:
-lpthread is an external library which is used to create threads in Ubuntu.
Commonly used along with c/c++ threaded programs. Usually it comes
preinstalled with Ubuntu or along with the gcc compiler. It seems you don't
have that library. Try installing the gcc compiler (which will most
probably contain the lpthread library) or try intalling it separately using
Ububtu Software center.

Regards,
Rohit

Athithyaa, 2015-08-31:
Yes, did you try updating the source.list file to get updated packages list
for ROS ?.

Installing ROS Indigo on Ubuntu 14.0.4.X was smooth sailing for me.

Try this http://wiki.ros.org/indigo/Installation/Ubuntu. It has clear cut
info.

Thanks,
Athithyaa

mira67, 2015-08-31:
ROS installation has no issue.
The installation of package for omni 60 has errors. Will troubleshoot
tomorrow.
Thanks.

Athithyaa, 2015-08-31:
Sorry, my bad.... :(
understood your issue.

A.

ryanleonard, 2015-09-01:
Hey everyone, 
So Regina and I are working on installing some things for the Tegra, but we aren't sure exactly where you guys left off. 

Can you confirm that ROS has/hasn't been installed?

We ask this because as far as we can tell, -lpthread has not yet been installed...

For future reference, we are leaving a text file under Documents on the tegra itself to let everyone else know what we have done so that if any problems arise in the future we can have some documentation which may help to resolve the issue.

mira67, 2015-09-01:
The driver package for omni60 was downloaded to jetson and under /Download folder.

installed image-transport, image-info-manager for ROS to support the camera
installed usblib which was missing
The same issue metioned previously occured when installing package for omni60 on jetson

If any of you have more experience on installing 3rd party package for ROS, the instructions for installing omni60 is in the readme file in the driver directory /Downloads/ros-indigosdk-2.0.0, you can help check the issue. Meanwhile, I am researching the solutions.

mira67, 2015-09-02:
Found the solution! Going to install the driver and write a usable instruction to wiki.

mira67, 2015-09-05:
Thanks to someone who helped fixed the problem.
"ARM_NEON" flag has to be defined to make thing work smoothly. 
Now the camera sdk/driver was installed and linked with ROS. DONE!
(Going to create another issue to track "UDEV RULE" for the Camera)

closed 2015-09-05 by mira67

rmaheshkumarblr, 2015-09-18:
Can you please provide instructions on how to start the Occam camera on the wiki?

mira67, 2015-09-18:
ok, will document that tomorrow