correll/advancedrobotics #15: Install OpenCV for Tegra

correll, 2015-08-27:


rmaheshkumarblr, 2015-08-30:
This link provides how to go about installing OpenCV for Tegra: http://elinux.org/Jetson/Installing_OpenCV
Will try it in the lab on Monday or Tuesday

correll, 2015-08-30:
That's good. Need to coordinate with whoever flashes the Jetson with LT4 21.x (Rohit?).

I recommend to go the route of using the precompiled option (optimized for tegra) and add support for sift/surf manually at a later point.

https://en.m.wikipedia.org/wiki/Speeded_up_robust_features

RoRa9362, 2015-08-30:
Yea. I'll be doing the Jetson Flashing on Monday. We will do the OpenCV
together.

Regards,
Rohit

rmaheshkumarblr, 2015-08-31:
Hi Rohit, 

I am in the Lab, let me know when you would be available. I don't have your number yet so I thought will use Github instead for the time being..

ikemole, 2015-09-01:
I have installed CUDA. According to the instructions link above, it is required to install CUDA on the Jetson before you can install OpenCV, so now you can go ahead and install OpenCV.

RoRa9362, 2015-09-02:
I have install OpenCV. Issue Closed!

closed 2015-09-02 by RoRa9362

RoRa9362, 2015-09-02:
I have installed the native-compiled OpenCV which has full functionality including GPU support.
Common way of running c++ programs in openCV is (running a sample program called edge.cpp, dont forget to include --cflags and --libs from opencv):
cd opencv-3.0.0/samples/cpp/
g++ edge.cpp -o edge $(pkg-config opencv --cflags --libs)
./edge