correll/advancedrobotics #16: Collect a list of BuffId + Names to obtain lab access

correll, 2015-08-27:


closed 2015-09-02 by correll

RohitRaje9362, 2015-10-23:
To @ryanleonard 
Do you have the existing list of BuffId + Names so that I can update it for access to the Correll lab?

rmaheshkumarblr, 2015-10-24:
There are no 80-20's in the lab and I have no access to Correll lab, so Zach could not get the hardware fixed.

In case you are collecting the name and buffid list again. Here are the details.

Name: Mahesh Kumar Ravindranathan
BuffID: 6014235005240075

correll, 2015-10-24:
Radhen was in the lab earlier. Shall I come in and open the door?

rmaheshkumarblr, 2015-10-24:
Rohit is on his way to the college, he has access. He must be able to open the door.

ryanleonard, 2015-10-24:
If you guys still have trouble getting in for whatever reason, just give me a call. 
720-339-6304
Also, Steve found a bag of some 80x20 stuff which is up against the wall by the jag. I have no idea whats in there though...

Date: Sat, 24 Oct 2015 12:42:10 -0700

jilu7883, 2015-10-24:
I am in the engineering center right now, I can open the door

rmaheshkumarblr, 2015-10-24:
@RohitRaje9362 opened the door.

RohitRaje9362, 2015-10-24:
I already did.

Rohit

jilu7883, 2015-10-24:
OK