correll/advancedrobotics #17: Install wireless mini PCIe card (need to be ordered)

correll, 2015-08-27:


correll, 2015-08-27:
There is a SMA (?) cable wiring an antenna to the outside. Need to make sure the PCI card supports external antenna.

rmaheshkumarblr, 2015-09-05:
The following link shows the PCI cards that are supported via the Grinch Kernal 

http://elinux.org/Jetson/Network_Adapters

The ones which we have are different from the supported list. Will try to use them anyways to check if they are detectable.

One of them we have is Intel Centrino 2230 which Nvidia plans to support in future for the Tegra platform. Reference from the link provided above.

correll, 2015-09-05:
If you find one on Amazon, I can buy it. Make sure it has a connector for the external antenna!

rmaheshkumarblr, 2015-09-05:
The recommended one for Jetson using the grinch kernal is Intel : 7260 HMW

Instruction on how to install and use: http://jetsonhacks.com/2014/10/12/installing-wireless-intel-7260-adapter-nvidia-jetson-tk1/

Amazon link for the module: http://www.amazon.com/gp/product/B00DMCVKMU/ ( It can have two external antennas attached to it )
Amazon link if antennas are required: http://www.amazon.com/Mini-Antenna-RP-SMA-Bulkhead-Pigtail/dp/B007XVHQ9M/

correll, 2015-09-05:
I'd be glad to order, but do we need the antenna or can use that already in the jaguar?

rmaheshkumarblr, 2015-09-05:
I think Jaquar came with just one antenna. We could use one of those and buy one more since this wireless module has slot for two external antennas.

correll, 2015-09-11:
I have ordered the module and antenna

rmaheshkumarblr, 2015-09-18:
We just fixed the wireless module and it started working. Looks like it got the drivers from the grinch kernel.

closed 2015-09-18 by rmaheshkumarblr