correll/advancedrobotics #18: Provide power / wiring to IMU

correll, 2015-08-27:
Needs 5V. Currently uses RS232, need to migrate to FTDI-USB cable

correll, 2015-09-09:
I bought a RS232 to USB adapter.

BryanBo-Cao, 2015-09-10:
Tutorial for SparkFun "9DOF Razor IMU": https://github.com/ptrbrtz/razor-9dof-ahrs/wiki/Tutorial.
I am trying to figure out a way to integrate it into Jaguar in both Hardware and Software perspectives.

correll, 2015-09-10:
I recommend following this tutorial until you are sure data is being received on the Jetson. I would then move to the ROS package that we discovered during the meeting.

BryanBo-Cao, 2015-09-12:
Via "USB-RS-232 Convertor” connecting laptop to IMU, the IMU had no power supply and therefore could not work.

After desoldering the pins on IMU from "USB-RS-232 Convertor”, we cut a USB cable and soldered the Red wire (+5V) to + pin and Black wire to - pin on IMU, with Orange wire (TXD) connected to RXI on IMU and Yellow wire (RXD) to TXD on IMU.
Reference:
[image]

Following the tutorial and having installed Razor AHRS Firmware and Arduino,  uploading firmware stucked like this
[image]
But the Serial Monitor indicates that my laptop could receive data:
[image]

Conclusion:
From the testing result we might conclude that we could receive data from IMU but cannot send signals to IMU.

Question:
In the tutorial, for wired the IMU uses  FTDI Basic Breakout - 3.3V to connect computer, which I can’t find in lab, but we just cut the USB cable and soldered the wires instead,  do we need to order one?

Thank @mcguire-steve @ryanleonard @RoRa9362 @Jorge-C for help~!

RohitRaje9362, 2015-09-14:
Where is the Sparkfun IMU? Im in the lab and cant seem to find it!

correll, 2015-09-15:
It should be mounted inside the robot

BryanBo-Cao, 2015-09-15:
Yes, it is just mounted right in the middle of the robot, with a USB cable connected--Just checked that in the lab.

BryanBo-Cao, 2015-09-15:
Tested today: same result, data received from IMU but cannot fully upload firmware to it.

The picture below displays how wires are connected to IMU:
[image]

RohitRaje9362, 2015-09-16:
I tested the IMU today evening. After certain detailed experimentation under the Digital Oscilloscope, testing the wire signals and their configurations. The USB-FTDI cable is perfectly working. 

The only problem is that the RTS from the USB (Active Low) has to be mapped to the DTR (Active High). Also, the CTS from ATmega386 (Active High) has to be mapped to the CTS of USB (Active Low). So, there needs to be inversion of signals for the RTS-CTS handshake to be successful. Only after this is successful can the transmission(upload) begin. Looking at the schematics of the Sparkfun FTDI 3.3V, it does have pins in the right configuration which does not need the extra inversion as is required with the current USB-FTDI cable.

I think we should just buy the SparkFun FTDI 3.3V.

[image]

Schematics of Sparkfun FTDI 3.3V: http://cdn.sparkfun.com/datasheets/BreakoutBoards/FTDI%20Basic-v22-3.3V.pdf

Sparkfun FTDI 3.3V: https://www.sparkfun.com/products/9873

Thanks to @Jorge-C

correll, 2015-09-16:
Great work. We should have a ton of the latter in the lab.

BryanBo-Cao, 2015-09-16:
Test:
The IMU is connected to FT232R breakout that was originally in the Robot:
    FT232R BREAKOUT <--> IMU
    GND                           <--> GND
    CTS                            <--> CTS
    3.3V                            <--> 3.3V
    TXO                            <--> RXI
    RXI                             <--> TXO
    DTR                            <--> DTR
FT232R is connected to laptop via a USB cable.

Setting of the Arduino software:
    Board: "Arduino Pro or Pro Mini"
    Processor: "ATmega328 (3.3V, 8 MHz)"
    Port: (select the one that connects to FT232R)
    Firmware: "Razor_AHRS.ino"
    Hardware selected in the firmware code: "HW__VERSION_CODE 10736"

Serial Monitor (Baud Rate: 57600) in Arduino could receive data from IMU, test with Digital Oscilloscope showed that signal could be sent from Serial Monitor to the pins in IMU.

Error Message:
Attempted to upload firmware to 9DOF Razor IMU but got error message from Arduino software console:
    "avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x24"...

Thanks to @RohitRaje9362 . Will troubleshoot today. 
BTW, does anyone have any ideas?

rmaheshkumarblr, 2015-09-16:
I have seen the sync error occurs when the reset button on the Arduino is not pressed in sync with the upload button on the Arduino software. Not sure if it is relavent here though.

Jorge-C, 2015-09-16:
Yeah, try uploading and resetting the board at the same time, or slightly shifted times. If we were using sparkfun's breaktout board it'd take care of that, but without it I think we manually need to reset as we upload, which is tricky to get right.

correll, 2015-09-16:
If that's the only problem, it should be an easy fix as the boot loader usually waits for a few seconds and we don't need to write to the imu that often.

ryanleonard, 2015-09-17:
So, it appears as though we have exhausted all of the options we have found on the internet for writing to the imu except for resetting the bootloader. Because we can receive input from the sensor, Sina suggests that we leave everything as it is, and manipulate the data we receive to solve for drifting or any other issues.

closed 2015-09-17 by correll

ryanleonard, 2015-09-18:
changed the imu firmware the firmware from:https://github.com/ptrbrtz/razor-9dof-ahrs/wiki/Tutorial.

Everything WORKS!
we can read and send commands to the imu. 
Thanks to Rohit, Bryan and Steve