correll/advancedrobotics #19: Provide power / wiring to GPS

correll, 2015-08-27:
Currently uses RS232, need to to migrate to FTDI232

rmaheshkumarblr, 2015-09-04:
The GPS Module that came with the Jaquar Platform is Garmin 18x GPS Navigator Unit

Specification Sheet: http://static.garmincdn.com/pumac/GPS_18x_Tech_Specs.pdf

[image]

The GPS 18x LVC/18x-5Hz interfaces to a serial port. The unit accepts TIA-232-F (RS-232) level inputs and transmits voltage levels from ground to the input voltage, TIA-232-F (RS-232) polarity.

rmaheshkumarblr, 2015-09-05:
RS232 to FTDI232 

Reference: http://www.usconverters.com/index.php?main_page=page&id=62

The Quick Checklist:

Preferred processor chip: FTDI or Silabs (very reliable)
Preferred serial driver chip: ZyWyn, Maxin or NI (very reliable)
Minimum 5.0VDC output power at serial connector (ensures reliable data transmission)
Build-in electro-static protection for at least 600W (protects against static electricity)
LED lights for TX and RX activity (nice to have when troubleshooting)

We could go for one of the following:

http://www.amazon.com/Gearmo%C2%AE-RS-232-Serial-Adapter-Chipset/dp/B004WL7MIQ

http://www.amazon.com/GearMo%C2%AE-Header-Like-FTDI-TTL-232R-5V/dp/B004LC28G2

http://www.amazon.com/gp/product/B00AHYJWWG/

correll, 2015-09-05:
I have ordered the first item. It might also work for the IMU.

rmaheshkumarblr, 2015-09-11:
The Garmin GPS by default had only 4 wires connected to the output terminal.
As per the Specification, there are suppose to be 6 wires. One wire which is going to PPS ( Pulse per second ) was not to be found in the output terminal. The other was a extra GND.

Steve suggested to use the PPS for enabling NTP using the Chrony software on top of Linus PPS ( This would enable all the robots to be in sync with respect to time since they would be using the time from the satellites via the GPS instead of using the local clock ).

I cut the wire of Garmin GPS Module and found 7 wires ( 3 GNDs , Vcc , RX , TD , PPS )

With the help of Steve, got the RS232 Connector (USB to Serial) to be opened as well. Later we removed connection to the pins 1 6 7 8 and 9. Manually soldered pin 1 to Vcc of the USB ( 5V ) and pin 6 to GND.

We connected the USB to Serial cable to Computer and we started getting inputs from the GPS module. Tried with different baud rate and figured out that it is currently set to 115200 .

[image]

Command to use while connecting from Linux:

[image]

[image]

correll, 2015-09-11:
Great job!

rmaheshkumarblr, 2015-09-15:
Entered udev rules for the GPS module to be linked via the name GPS_System:

ubuntu@tegra-ubuntu:~$ cat /etc/udev/rules.d/10-self-defined.rules | head -1
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AL00ITFV" ,  MODE="0666" , SYMLINK+="GPS_System"
ubuntu@tegra-ubuntu:~$

ubuntu@tegra-ubuntu:~$ ls -ltr /dev/GPS_System 
lrwxrwxrwx 1 root root 7 Sep 15 19:15 /dev/GPS_System -> ttyUSB0
ubuntu@tegra-ubuntu:~$ 

ubuntu@tegra-ubuntu:~$ screen /dev/GPS_System 115200n81

$GPGGA,001908.0,4000.41520,N,10515.80242,W,0,00,,,M,,M,,*5B
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPVTG,,T,,M,,N,,K*4E
$GPRMC,001908.2,V,4000.41520,N,10515.80242,W,,,220899,008.7,E*62
$GPGGA,001908.2,4000.41520,N,10515.80242,W,0,00,,,M,,M,,*59

Method to use UDEV ( Collecting the required Information ):

ubuntu@tegra-ubuntu:~$ lsusb
Bus 002 Device 003: ID 285e:3efd  
Bus 002 Device 002: ID 2109:8110  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC

ubuntu@tegra-ubuntu:~$ dmesg | tail -5
[   45.303606] usb 1-3.3: Number of endpoints 2
[   45.303613] usb 1-3.3: Endpoint 1 MaxPacketSize 64
[   45.303619] usb 1-3.3: Endpoint 2 MaxPacketSize 64
[   45.303625] usb 1-3.3: Setting MaxPacketSize 64
[   45.304351] usb 1-3.3: FTDI USB Serial Device converter now attached to ttyUSB0
ubuntu@tegra-ubuntu:~$ 

[   45.303606] usb 1-3.3: Number of endpoints 2
[   45.303613] usb 1-3.3: Endpoint 1 MaxPacketSize 64
[   45.303619] usb 1-3.3: Endpoint 2 MaxPacketSize 64
[   45.303625] usb 1-3.3: Setting MaxPacketSize 64
[   45.304351] usb 1-3.3: FTDI USB Serial Device converter now attached to ttyUSB0
ubuntu@tegra-ubuntu:~$ 

ubuntu@tegra-ubuntu:~$ udevadm info -a -p $(udevadm info -q path -n /dev/ttyUSB0)

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/platform/tegra-xhci/usb1/1-3/1-3.3/1-3.3:1.0/ttyUSB0/tty/ttyUSB0':
    KERNEL=="ttyUSB0"
   <b> SUBSYSTEM=="tty"</b>
    DRIVER==""

  looking at parent device '/devices/platform/tegra-xhci/usb1/1-3/1-3.3/1-3.3:1.0/ttyUSB0':
    KERNELS=="ttyUSB0"
    SUBSYSTEMS=="usb-serial"
    DRIVERS=="ftdi_sio"
    ATTRS{port_number}=="0"
    ATTRS{latency_timer}=="1"

  looking at parent device '/devices/platform/tegra-xhci/usb1/1-3/1-3.3/1-3.3:1.0':
    KERNELS=="1-3.3:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="ftdi_sio"
    ATTRS{bInterfaceClass}=="ff"
    ATTRS{bInterfaceSubClass}=="ff"
    ATTRS{bInterfaceProtocol}=="ff"
    ATTRS{bNumEndpoints}=="02"
    ATTRS{supports_autosuspend}=="1"
[... 99 lines cut ...]
    ATTRS{idProduct}=="0002"
    ATTRS{bDeviceClass}=="09"
    ATTRS{product}=="Nvidia xHCI Host Controller"

  looking at parent device '/devices/platform/tegra-xhci':
    KERNELS=="tegra-xhci"
    SUBSYSTEMS=="platform"
    DRIVERS=="tegra-xhci"

  looking at parent device '/devices/platform':
    KERNELS=="platform"
    SUBSYSTEMS==""
    DRIVERS==""

ubuntu@tegra-ubuntu:~$

ubuntu@tegra-ubuntu:~$ cat /etc/udev/rules.d/10-self-defined.rules | head -1
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AL00ITFV" ,  MODE="0666" , SYMLINK+="GPS_System"
ubuntu@tegra-ubuntu:~$ 

Applying Changes: 
( NOTE : Make sure you have only one line per device, the complete entry must be made in a single line. )

ubuntu@tegra-ubuntu:~$ sudo udevadm control --reload-rules && sudo service udev restart && sudo udevadm trigger
ubuntu@tegra-ubuntu:~$

rmaheshkumarblr, 2015-09-16:
Installing robot_localization package using the link : http://wiki.ros.org/robot_localization

I had to install roslint as it was a dependency: 
 (Reference Link: http://answers.ros.org/question/204078/roslint-error/ )

I had to install nmea_navsat_driver-package:
(Reference Link : http://answers.ros.org/question/190264/trouble-installing-nmea_navsat_driver-package/ )

Created a launch file.

ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ ls
GPS_System.launch  ekf_template.launch  navsat_transform_template.launch  ukf_template.launch
ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ cat GPS_System.launch 
<launch>

    <node pkg="nmea_navsat_driver" type="nmea_serial_driver" name="navsattest" respawn="true">
       <param name="port" value="/dev/GPS_System"/>
       <param name="baud" value="115200"/>
    </node>
    
</launch>

ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ pwd
/home/ubuntu/catkin_ws/src/robot_localization/launch
ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ 

ubuntu@tegra-ubuntu:~/catkin_ws$ rosnode info /navsattest
--------------------------------------------------------------------------------
Node [/navsattest]
Publications: 
 * /fix [sensor_msgs/NavSatFix]
 * /rosout [rosgraph_msgs/Log]
 * /vel [geometry_msgs/TwistStamped]
 * /time_reference [sensor_msgs/TimeReference]

Subscriptions: None

Services: 
 * /navsattest/get_loggers
 * /navsattest/set_logger_level

contacting node http://tegra-ubuntu:49325/ ...
Pid: 5431
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS

ubuntu@tegra-ubuntu:~/catkin_ws$ 

ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ rostopic echo /fix
header: 
  seq: 28
  stamp: 
    secs: 1442445792
    nsecs: 31938076
  frame_id: /gps
status: 
  status: -1
  service: 1
latitude: 40.00692
longitude: -105.263373667
altitude: nan
position_covariance: [nan, 0.0, 0.0, 0.0, nan, 0.0, 0.0, 0.0, nan]
position_covariance_type: 1
---

ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ rostopic echo /time_reference 
header: 
  seq: 1
  stamp: 
    secs: 1442445840
    nsecs: 637577056
  frame_id: /gps
time_ref: 
  secs: 1442364563
  nsecs: 0
source: /gps
---
header: 
  seq: 2
  stamp: 
    secs: 1442445840
    nsecs: 836549997
  frame_id: /gps
time_ref: 
  secs: 1442364563
  nsecs: 0
source: /gps

closed 2015-09-17 by correll