correll/advancedrobotics #20: Implement FTDI management in Linux

correll, 2015-08-27:
FTDI devices are labeled as /dev/ttyUSBx. We will need to provide fixed names (via the devices serial numbers) in order to prevent confusion.

rmaheshkumarblr, 2015-09-16:
ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ cat /etc/udev/rules.d/10-self-defined.rules  | head -2
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AL00ITFV" ,  MODE="0666" , SYMLINK+="GPS_System"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AH02LWD6" ,  MODE="0666" , SYMLINK+="IMU_System"
ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ 

ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ ls -ltrh /dev/GPS_System /dev/IMU_System 
lrwxrwxrwx 1 root root 7 Sep 16 23:32 /dev/GPS_System -> ttyUSB0
lrwxrwxrwx 1 root root 7 Sep 16 23:32 /dev/IMU_System -> ttyUSB1
ubuntu@tegra-ubuntu:~/catkin_ws/src/robot_localization/launch$ 

Bus 001 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 005: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC

Jorge-C, 2015-09-21:
This can be closed, right?

rmaheshkumarblr, 2015-09-21:
I think we will have to continue with updating the udev rules till all the hardwares are connected to ROS.
Right now only Occam Camera, GPS and IMU are updated. We could close this since the FTDI devices are updated but we could open one new issue for all devices in general ( DC Motors, IMU from stuart platform, etc.. )

rmaheshkumarblr, 2015-10-13:
Updated the UDEV rules to include the USB module that connects to the Motor Controller.
Also modified an existing 3.3V FTDI module to work with IMU so that Motor Controller and IMU can work independently (Not sharing the same FTDI cable anymore)
```
/etc/udev/rules.d/10-self-defined.rules

SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AL00ITFV" ,  MODE="0666" , SYMLINK+="GPS_System"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="A703EBFG" ,  MODE="0666" , SYMLINK+="IMU_System"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AH02LWD6" ,  MODE="0666" , SYMLINK+="MOTOR_System"
SUBSYSTEM=="usb", ATTRS{idVendor}=="285e" , ENV{ID_SERIAL_SHORT}=="811168f166914178", MODE="0666" , SYMLINK+="OccamCamera"
```

rmaheshkumarblr, 2015-10-15:
Updated udev rules to detect joystick automatically.

Udev rules will  automatically change the baud rate to the motor control system to 115200 by default.

```
ubuntu@tegra-ubuntu:~/catkin_ws$ cat /etc/udev/rules.d/10-self-defined.rules
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AL00ITFV" ,  MODE="0666" , SYMLINK+="GPS_System"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="A703EBFG" ,  MODE="0666" , SYMLINK+="IMU_System"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403" , ATTRS{manufacturer}=="FTDI"  ,ATTRS{serial}=="AH02LWD6" ,  MODE="0666" , SYMLINK+="MOTOR_System" , RUN+="/bin/stty -F /dev/MOTOR_System 115200"
KERNEL=="js*" , SUBSYSTEM=="input", ATTRS{idVendor}=="046d" , ATTRS{manufacturer}=="Logitech" , ATTRS{serial}=="74469186" , MODE="0666" , SYMLINK+="JOYSTICK_System"
SUBSYSTEM=="usb", ATTRS{idVendor}=="285e" , ENV{ID_SERIAL_SHORT}=="811168f166914178", MODE="0666" , SYMLINK+="OccamCamera"
ubuntu@tegra-ubuntu:~/catkin_ws$ 

```

Jorge-C, 2015-11-25:
The udev rules for the stewart platform components have also been integrated.

closed 2015-11-25 by Jorge-C