correll/advancedrobotics #22: Add USB3 hub

correll, 2015-08-27:
The hub needs to be mounted to the robot somehow. Maybe laser-cut a fixture?

rcr2f, 2015-10-02:
Mount in one of these slots?

[image]

rcr2f, 2015-10-02:
maybe something like these?

http://www.digikey.com/product-search/en?pv28=872&pv392=1677&FV=fff40016%2Cfff802e4&k=waterproof+connector&mnonly=0&newproducts=0&ColumnSort=0&page=1&quantity=0&ptm=0&fid=0&pageSize=25

ryanleonard, 2015-10-02:
http://www.l-com.com/usb-usb-type-a-waterproof-field-installable-connector

and cap:

http://www.l-com.com/firewire-din-scsi-sata-waterproof-cap-for-ieee-1394-firewire-usb?cmp=PAT