correll/advancedrobotics #23: Restore battery packs.

correll, 2015-08-27:
Unfortunately, the batteries (LiPo) broke. Need to order new packs of follow instructions here http://www.electricrcaircraftguy.com/2014/10/restoring-over-discharged-LiPos.html

correll, 2015-08-29:
The packs are actually composed of four separate sections, so we can replace just
the individual parts that actually need replacing.

http://www.hobbyking.com/hobbyking/store/__24215__Turnigy_nano_tech_5000mah_3S_65_130C_Lipo_Pack_US_Warehouse_.html

Thanks to Steve McGuire.

RoRa9362, 2015-08-30:
I also heard you mention that only one of the cells was damaged. So, we
could actually just buy one cell and replace it.
But before that, we could try to recover that particular cell by Low Charge
Current in NiMH charger configuration.

http://www.hobbyking.com/hobbyking/store/__18560__Turnigy_5000mAh_1S_20C_Lipoly_Single_Cell_.html

This one seems to be quite a similar cell of the 3s1p Battery, only the
constant Discharge is 20C instead of 65C

Regards,
Rohit

correll, 2015-08-30:
I've read that repairing cells under 1V/cell is very risky as they can catch fire. I simply bought new ones.

Until then we can do anything involving the computer and sensors.

mcguire-steve, 2015-09-11:
Investigated yesterday - the recovery at low charge current appears to have worked. Pack reassembled and ready for service.

closed 2015-09-11 by mcguire-steve

correll, 2015-09-11:
Awesome. I could not get it to work with the lowest settings. How do we
prevent the new packs I bought from dying?