correll/advancedrobotics #4: Connect motor driver to joystick velocity screw

correll, 2015-08-27:


mira67, 2015-09-25:
1.      What's baud-rate required for the DSP motor sensing board in order to communicate with ROS/PC?

The baud-rate is 115200.
 
2.      What are the serial commands required for the DSP board in order to control the motors? Any detailed document?

We provide the protocol document and ROS sample code. The “DrRobotMotionSensorDriver” includes all the API functions for communication and control commands.
 
You also could refer to attached communication protocol document.
 
3.      We are not sure what's the firmware version on the current V4 robot, any new firmware we need to update to make it work with ROS?

You don’t need update the firmware on the robot. We provide the different version ROS sample code for different firmware. Please try the ROS sample code which could download from http://www.drrobot.com/files_2015/DrRobotJaguarv2.zip.

The communication protocol manual is in the email to google group, check it out.

closed 2015-09-29 by Branden-Romero

correll, 2015-09-29:
Great!

BryanBo-Cao, 2015-09-30:
First launch indoors from 1B15 to Correll Lab
[#1Video](https://youtu.be/mu-eZ2YVPoQ)
[#2Video](https://youtu.be/ZGt-fzrt39A)