correll/advancedrobotics #5: Attach Stewart platform to Jaguar

correll, 2015-08-27:


BryanBo-Cao, 2015-09-05:
Does anyone know where Stewart platform is in the lab? I could not find it in 1B15. Thanks!

rmaheshkumarblr, 2015-09-05:
If I am not wrong, I think it is in Correll's Lab: 1B21

correll, 2015-09-05:
It's in 1B21 on Andy McEvoy's desk.

correll, 2015-09-05:
Before attaching it, we should probably finish up the code for it...

I'll be in on Tuesday and can go over where we are with the platform.

Andy McEvoy

Research Assistant
Correll Lab
Department of Computer Science
University of Colorado at Boulder

(832) 439 - 6150

rcr2f, 2015-10-02:
Update from the recent meetings:

Steve suggested we should use 80/20 rail on the Jaguar rather than punching a bunch of holes in the top plate. The Husky robot is an example of how we should setup mounting. We also shouldn't have to order anything since there is extra in the lab.

http://www.8020.net/t-slot-5.asp

[image]
[image]
[image]

rcr2f, 2015-10-02:
More ideas on the 80/20....

[image]

ryanleonard, 2015-10-15:
I just cut the railing, drilled holes for the screws, and hollowed out enough metal on the top layer to allow for the head of the screw to lay flush with the center of the railing. 

Professor Correll, Can you order some #7, 1.5in drywall screws so that we can attach the railing to the top plate?

correll, 2015-10-15:
Hi Ryan

can you find this online and send me a link? We can also pick this up at
McGuckin.

Nikolaus

correll, 2015-10-15:
We also have machine screws and washers here

ryanleonard, 2015-10-16:
I picked up some screws last night, and just finished cutting the crossbars. Now we need to get some attachments compatible with 20mm x 20mm 8020. Everything that I could find in the lab was too big and wouldnt fit the railing.

correll, 2015-10-16:
Let me know the mcmaster parts and i will order

Zolkin1, 2015-10-19:
[image]

Above is my current design for mounting the stewart platform based on what we discussed on Thursday. I raised the stewart platform 4 in above the jaguar top, that way the laser sensor can fit underneath. The laser sensor has three holes that should be able to mount into the 2 pieces of 80/20 in between the bottom of the stewart platform. I was not able to find the dimensions of the holes or any part of the laser sensor so the pieces of 80/20 might need to be shifted in a little. I am still trying to figure out how to mount the occam in the stewart platform while keeping it stable.

Zolkin1, 2015-10-23:
I can come in at 1pm on Saturday to work on the mounts.

closed 2015-11-23 by BryanBo-Cao