correll/advancedrobotics #9: Install RTABMAP

correll, 2015-08-27:


RoRa9362, 2015-08-30:
I will be installing RTABMAP on Tuesday. If ROS and OpenCV are installed on early Monday, I will be doing it that day itself. Do join me if you are free then?

Rohit

ryanleonard, 2015-08-30:
I'll be there Tuesday!Ryan

https://github.com/correll/advancedrobotics/issues/9#issuecomment-136155411

mira67, 2015-08-30:
I will try to be there 1130 Monday, and maybe we can install the omni 60
driver also.

Qi

jilu7883, 2015-08-31:
I can join in Tuesday, when you guys will meet?

ryanleonard, 2015-08-31:
Any time before 2 for me
Ryan

https://github.com/correll/advancedrobotics/issues/9#issuecomment-136517348

jilu7883, 2015-08-31:
I can be here around 10:15 tomorrow, maybe early

RoRa9362, 2015-09-01:
I will be there in the Lab on Tuesday after 1pm. So, you guys can go ahead if you arrive early. BTW, today (Monday), we finished flashing the Jettson (16GiB filesystem) and installed ROS indigo on it. We tried getting the PCIe wifi adapter to function but seems as if the necessary drivers are missing.

correll, 2015-09-01:
The PCIe will probably require the grinch kernel.

https://devtalk.nvidia.com/default/topic/823132/-customkernel-the-grinch-21-3-4-for-jetson-tk1-developed/

Do you have such a wireless card to play with?

RoRa9362, 2015-09-01:
There were two PCIe wireless cards already there in the Lab. Found by
Steve! One was from RTL(RealTek) and the other was Intel Centrino
Wireless-N 2230.

correll, 2015-09-01:
I couldn't actually get this to work previously. I could see the interface and wireless networks, but it would not connect. Maybe one needs to hook up the antenna.

rmaheshkumarblr, 2015-09-01:
We could get the realtek one working installing the drivers but the problem was that the system hanged after a certain amount of time because of some issue. In the error messages we could see something mentioned about wpa_supplicant.

correll, 2015-09-01:
Can you post the whole error message? Maybe someone can figure it out.

RoRa9362, 2015-09-02:
Installed the grinch kernel. Will check the RTL PCIe again tomorrow!

closed 2015-09-17 by correll