correll/advancedrobotics #28: Document drivetrain with schematic diagram

correll, 2015-09-09:


mira67, 2015-09-24:
[image]

We can start based on this rough diagram to add new components in

jilu7883, 2015-09-24:
Did anyone work one the hardware yet? to run the batteries devices

jilu7883, 2015-09-24:
http://jaguar.drrobot.com/images/Jaguar_V2_Manual.pdf

All-Terrain Autonomous NavigatIon Robot with GPS-IMU

correll, 2015-09-24:
Lets use open office to update this. We are now using USB and the Jetson
board instead of just sending everything over wireless.

mira67, 2015-09-24:
Motor driver board: Sabertooth dual 25A motor driver

http://www.trossenrobotics.com/store/p/5466-Sabertooth-dual-25A-motor-driver.aspx?feed=Froogle&gclid=Cj0KEQjwvo6wBRCG3Zv92ZSLlIYBEiQA5PLVAgbGphlm5MaYKpGL99ePjuPSq-RCBoKh8Cz2zYwVtZIaArnA8P8HAQ