correll/advancedrobotics #31: Flash right IMU firmware...

Jorge-C, 2015-09-25:
...using sketch from https://github.com/KristofRobot/razor_imu_9dof/tree/indigo-devel/src/Razor_AHRS (it is slightly different from the one [currently flashed](https://github.com/ptrbrtz/razor-9dof-ahrs/blob/master/Arduino/Razor_AHRS/Razor_AHRS.ino)).

Jorge-C, 2015-09-25:
We ended up having to use atmel to upload new firmware and the sketch. Commands necessary for calibration work! Thanks @RohitRaje9362 @rmaheshkumarblr

rmaheshkumarblr, 2015-12-12:
The problem was with the IMU. The IMU is not responding. We probably need to flash the IMU again. Tried flashing using Arduino and it does not seem to work because of the sync error.

RohitRaje9362, 2015-12-12:
Tried flashing the Boot-loader and Firmware inside the Jaguar IMU. 
The Bootloader is flashed successfully using the Atmel ICE programmer as given in the video in the following link:
https://startingelectronics.org/tutorials/arduino/arduino-uno-burning-bootloader-avr-studio/

However, we get 'programmer not in sync: 0x00' error while uploading the firmware for the IMU using Arduino IDE. Instructions are given here:
https://github.com/ptrbrtz/razor-9dof-ahrs/wiki/Tutorial

Note that this procedure was working previously but doesn't seem to work now. Can't yet figure out what's the problem!

We also found out that the Razor IMU for the stewart platform is being fed 5V VCC power through the FTDI into the 3.3V port of IMU. That's the setup that Andy has had and it has always worked. So, we tried the same configuration on the Jaguar IMU.
And, we also tried the alternate configuration i.e. to reroute the Power and Ground wires from the Adafruit FTDI since they are 5V, while the others (TXD, RXD, DTR, CTS) are 3.3V.

Thanks @Jorge-C

correll, 2015-12-12:
Is it the same IMU? We do have 3.3V FTDI cables.

rmaheshkumarblr, 2015-12-12:
It is the same IMU. It does not seem to work. Out of the two 3.3V FTDI cables that we got from the lab. One of them is not being detected by the laptop/Tegra as a FTDI cable / USB. The other one provides a voltage of around 2.6V~2.7V instead of 3.3V. I even powered 3.3V directly and used the previously used FTDI and it still did not work.