correll/advancedrobotics #32: Calibrate IMU

Jorge-C, 2015-09-25:
Depends on #31

BryanBo-Cao, 2015-09-25:
In the Razor_AHRS.ino, under 'SENSOR CALIBRATION' find the section that reads 'Magnetometer (extended calibration)'
Replace the existing 3 lines with these:
```
#define CALIBRATION__MAGN_USE_EXTENDED true
const float magn_ellipsoid_center[3] = {125.243, -34.6189, -104.951};
const float magn_ellipsoid_transform[3][3] = {{0.875048, 0.0736508, -0.00920915}, {0.0736508, 0.863432, 0.00348322}, {-0.00920915, 0.00348322, 0.999281}};
```

Jorge-C, 2015-09-25:
For the record, we had to use an old version from 2013 ([v0.24](http://code.google.com/p/efficient-java-matrix-library/downloads/detail?name=ejml-0.24.jar&can=4&q=)) of the EJML library to get magnetometer calibration to work.

BryanBo-Cao, 2015-09-25:
[magnetom.float.txt](https://github.com/correll/advancedrobotics/files/2751/magnetom.float.txt)

@Jorge-C

Jorge-C, 2015-09-25:
Thanks! Here is the calibration config file
[my_razor.yaml.txt](https://github.com/correll/advancedrobotics/files/2753/my_razor.yaml.txt)

Note that the extension `.txt` has to be removed from all the attached files.

BryanBo-Cao, 2015-09-25:
Following the instructions on [7.1.3 Calibrating the magnetometer](http://wiki.ros.org/razor_imu_9dof), [Processing 2.2.1] (https://processing.org/download/?processing) is downloaded on Mac (Version Yosemite 10.10.5).

**Add jar library**: download 2013 ([v0.24](https://code.google.com/p/efficient-java-matrix-library/downloads/detail?name=ejml-0.24.jar&can=4&q=)) and change its name "ejml-0.24.jar" to "EJML.jar" and copy it to the directory: "Document/Processing/libraries/EJML/library/EJML.jar". After this Processing needs to be re-open.

Before running the code "Magnetometer_calibration.pde" on Processing, you needs to find the port connecting to IMU and assign it by  "final static int SERIAL_PORT_NUM = {$(your port number)}".

Re-open Processing and below is what it looks on my Mac:
[image]

Thank @Jorge-C @rmaheshkumarblr for help!

closed 2015-09-25 by BryanBo-Cao