correll/advancedrobotics #35: Calibrate the IMU for the Stewart Platform

rcr2f, 2015-09-30:
Is someone familiar with calibrating the Sparkfun Razor 9DOF IMU already and want to take this on? 

https://github.com/ptrbrtz/razor-9dof-ahrs/wiki/Tutorial

BryanBo-Cao, 2015-09-30:
@Jorge-C and I have calibrated the IMU but outside the robot. But the magnetometer of the IMU needs to be calibrated again since it is affected by other metals and electronic components.

Jorge-C, 2015-09-30:
Bryan, there's another IMU in the Stewart platform. That one needs to have the magnetometer calibrated (Andy calibrated accels and gyros already, but it'd still be good to calibrate them again).

BryanBo-Cao, 2015-10-22:
Below is the calibration config file for accelerometer and gyro.
[razor_stewart.yaml.txt](https://github.com/correll/advancedrobotics/files/18952/razor_stewart.yaml.txt)
Magnetometer could be calibrated when the IMU is attached to the steward platform.

Note that the extension .txt has to be removed.
Thank @Jorge-C and @RohitRaje9362 for help.

closed 2015-11-15 by Jorge-C