correll/advancedrobotics #38: Power for the external sensors and stewart platform

rcr2f, 2015-10-02:
3 power output ports:

RAW / 12V 5A / 5V 1A
CPC (Circular Plastic Connectors)

Also, need to find a place to put these.

rmaheshkumarblr, 2015-10-02:
I think this would be the correct one for power. Need to check with Steve.

http://www.amazon.com/Wellsem-Waterproof-Dustproof-Aviation-Connector/dp/B0159TQ6JK

[image]

correll, 2015-10-02:
Please repost if it is the right one and I will add it to the order

rcr2f, 2015-10-02:
More options...We will meet with Steve early next week to figure out the right parts.

http://www.mouser.com/Connectors/Circular-Connectors/Standard-Circular-Connectors/Standard-Circular-Connector/CPC-Series/_/N-av49j?P=1z0yj6eZ1yvsm2d

mcguire-steve, 2015-10-06:
The existing holes in the robot are 1" diameter, basically shell size 13. I recommend these connectors:
TE 211401-1 (robot side): qty 2
Pins for above: 66591-1, qty 20

Assuming 18ga-22ga wiring:
TE 211399-1 (cable side): qty 4
Sockets for above: 66592-1, qty 40

It's a 7-pos connector, so we can include grounds for each power line (BATT RAW, 12V, 5V) tied internally. Each pin is rated for 13A of current, which is even enough to power the Kinova arm as an external device.

rmaheshkumarblr, 2015-10-06:
TE 211401-1 (robot side): qty 2
http://www.amazon.com/TE-Connectivity-Part-Number-211401-1/dp/B00KTX6B4M/  ( 2 of these )
Pins for above: 66591-1, qty 20
http://www.amazon.com/Connectivity-66591-1-TYPE-CONT-M-MATE/dp/B00DED5500/  (Each one has 2 pins hence we need to order 10 of these)

TE 211399-1 (cable side): qty 4
http://www.amazon.com/TE-Connectivity-Part-Number-211399-1/dp/B00KTX8GQI/ ( 4 of these )
Sockets for above: 66592-1, qty 40
http://www.amazon.com/TE-Connectivity-Part-Number-66592-1/dp/B00KSUKK3O/ ( 40 of these )

correll, 2015-10-06:
I've put these and Rebecca's orders in.

mcguire-steve, 2015-10-13:
I forgot a connector cover for the robot side and the backshells for the external power cable:
Connector cover:
TE 211870-2 (qty 2)
Backshell: 
TE 206966-7 (qty 4)

rmaheshkumarblr, 2015-10-15:
TE 211870-2 (qty 2)
http://www.amazon.com/TE-Connectivity-211870-2-SEALING-RECEPT/dp/B00DECRN88/   -- 2 numbers
TE 206966-7 (qty 4)
http://www.amazon.com/TE-CONNECTIVITY-AMP-206966-7-THERMOPLASTIC/dp/B011COZROQ/ -- 4 numbers