correll/advancedrobotics #39: Implement SLAM with Occam60

BryanBo-Cao, 2015-10-05:
Occam60 can be open by entering '/home/ubuntu/occamdrivers/indigosdk-2.0.34/build/bin/read_images_opencv' on terminal shown below: 

[image]
Thank @ikemole !

But based on my understanding it looks like that RTAB-Map does not have an option to use Occam60 directly, (e.g. it can use Kinect directly). Error message is shown below:

[image]

Jorge-C, 2015-10-05:
Stitching doesn't seem to work (there's an image channel called `stitched_image0` which can be selected) but the displayed image is a long gray gradient:
[image]

Any ideas? BTW, I had to make some small code edits to get channel-changing working in Occam's demo.

mcguire-steve, 2015-10-05:
Stitching won't work until the internal camera calibration parameters are written by the Occam Windows calibration utility.

Jorge-C, 2015-10-05:
Thanks, created #40 to track that.

Steve also suggested getting started with monocular SLAM using [LSD-SLAM](http://vision.in.tum.de/research/vslam/lsdslam).

mira67, 2015-10-22:
Tutorial to setup LSD-SLAM with ROS: https://github.com/tum-vision/lsd_slam

correll, 2015-10-22:
Cool!

jilu7883, 2015-10-23:
we try to install LSD-SLAM, the errors shows below, we are thinking that may cause by there is two openCV versions.
that's the step we run:
```
sudo apt-get install ros-indigo-libg2o ros-indigo-cv-bridge liblapack-dev libblas-dev freeglut3-dev libqglviewer-dev libsuitesparse-dev libx11-dev
git clone https://github.com/tum-vision/lsd_slam.git lsd_slam
rosmake lsd_slam
```
(2.2 ROS indigo + Ubuntu 14.04)

Error message:
```
 main_stitchVideos.cpp:(.text+0x7dd4): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x7dfc): undefined reference to `cv::imread(cv::String const&, int)'
  main_stitchVideos.cpp:(.text+0x7e02): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x7e24): undefined reference to `cv::imread(cv::String const&, int)'
  main_stitchVideos.cpp:(.text+0x7e2a): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x7e4c): undefined reference to `cv::imread(cv::String const&, int)'
  main_stitchVideos.cpp:(.text+0x7e52): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x7e74): undefined reference to `cv::imread(cv::String const&, int)'
  main_stitchVideos.cpp:(.text+0x7e7a): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8272): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x829e): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
  main_stitchVideos.cpp:(.text+0x82ba): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8306): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8332): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
  main_stitchVideos.cpp:(.text+0x8340): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x838c): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x83b8): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
  main_stitchVideos.cpp:(.text+0x83c6): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x83d4): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8400): undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
  main_stitchVideos.cpp:(.text+0x8406): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8b42): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8b58): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8b6e): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8b86): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8b9c): undefined reference to `cv::String::allocate(unsigned int)'
  main_stitchVideos.cpp:(.text+0x8daa): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8df6): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8e06): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8e16): undefined reference to `cv::String::deallocate()'
  main_stitchVideos.cpp:(.text+0x8e42): undefined reference to `cv::String::deallocate()'
  CMakeFiles/videoStitch.dir/src/main_stitchVideos.cpp.o:main_stitchVideos.cpp:(.text+0x8e52): more undefined references to `cv::String::deallocate()' follow
  collect2: error: ld returned 1 exit status
  make[3]: *** [../bin/videoStitch] Error 1
  make[3]: Leaving directory `/home/ubuntu/catkin_ws/src/lsd_slam/lsd_slam_viewer/build'
  make[2]: *** [CMakeFiles/videoStitch.dir/all] Error 2
  make[2]: Leaving directory `/home/ubuntu/catkin_ws/src/lsd_slam/lsd_slam_viewer/build'
  make[1]: *** [all] Error 2
  make[1]: Leaving directory `/home/ubuntu/catkin_ws/src/lsd_slam/lsd_slam_viewer/build'
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/ubuntu/.ros/rosmake/rosmake_output-20151023-172505/lsd_slam_viewer/build_output.log
[rosmake-0] Finished <<< lsd_slam_viewer [FAIL] [ 7.74 seconds ]                                                   
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete. 
[ rosmake ] Results:                                                                                               
[ rosmake ] Built 41 packages with 1 failures.                                                                     
[ rosmake ] Summary output to directory                                                                            
[ rosmake ] /home/ubuntu/.ros/rosmake/rosmake_output-20151023-172505  
```

mira67, 2015-11-12:
Please use the following updated LSD_SLAM package to install, which was successfully installed on Ubuntu 14 + Indigo + OpenCV 2.4.8: https://github.com/mira67/lsd_slam
This updated version fixed the previous "cv...undefined" error. It will take >3min to build.

jilu7883, 2015-12-03:
I tried install LSD SLAM for the camera, for the command, i need the infotopic and calibration file for the Occam Camera, Did anyone know anything about that?
[image]

ikemole, 2015-12-03:
The package is called "occam_publisher" and the topic is called "occam\image". I don't think there is any calibration file for it. The publisher works without it.

jilu7883, 2015-12-03:
Thank U!