correll/advancedrobotics #40: Calibrate Occam60

Jorge-C, 2015-10-05:
:warning: Windows machine needed!

Follow instructions [in Occam's website](http://occamvisiongroup.com/knowledgebase/calibrating-omni-60/). You'll need to login (username `correll@correll.edu` and password the Jetson's should work).

There are checkerboards in the lab, so before printing and gluing another please check with Steve/Andy to borrow one.

Any takers?

--

To check the calibration, run the following in the Jetson:
```
$ cd ~/occamdrivers/indigosdk-2.0.34/build/
$ ./bin/read_calib
```
and you should get something like this with different numbers:
```
...
sensor 2:
  width = 752
  height = 480
  D = [[0,0,0,0,0]]
  K = [[1,0,0];[0,1,0];[0,0,1]]
  R = [[1,0,0];[0,1,0];[0,0,1]]
  T = [[0,0,0]]
...
```

RohitRaje9362, 2015-10-06:
I downloaded the Occam Indigo Software and tried calibration. However, not yet satisfied fully with the calibration. Need to do it one more time so that the stitched image is perfect.

Model and Brief Summary of Calibration are as follows:
[image]
[image]
 
[Caliberation Summary.txt](https://github.com/correll/advancedrobotics/files/8195/Caliberation.Summary.txt)

Something to Notice: In the Calibration Model, one of the sensor shows a more wide elongated distribution. That doesn't look normal! Any Comments on that?

Thanks to @BryanBo-Cao

BryanBo-Cao, 2015-10-07:
Attempted to calibrate today, based on what Steve suggested, for each sensor, we swiped the checkerboard from left to right, up to down along with the edge of the frame. On the four corners of the frame, we rotated respect to x, y axis and moved forwards and backwards.

Between 2 sensors we added more snapshots by moving the checkerboard up and down, forwards and backwards.

It took 1 and half hours to run the calibration on my desktop but the result did not look better. Some blurred areas could be seen between two sensors.

Thank @ryanleonard for help.

[image]

BryanBo-Cao, 2015-10-08:
# Calibration results

[Calibration Summary 20151007Wed.txt](https://github.com/correll/advancedrobotics/files/9821/Calibration.Summary.20151007Wed.txt)
[Calibration Detail 20151007Wed.txt](https://github.com/correll/advancedrobotics/files/9820/Calibration.Detail.20151007Wed.txt)

According to Step 7: Inspect results in the instructions on [Calibrating the Omni 60] (http://occamvisiongroup.com/knowledgebase/calibrating-omni-60/), in which it says "There are also ten-percentile reprojection errors given for the final reconstruction. If you have selected a good set of snapshots, typical results are less than 0.5 pixels with a distribution such as [[0.000847245,0.0341882,0.0626213,0.0929949,0.122866,0.153567,0.196825,0.243231,0.289278,0.363943]]."

Here's what we got from [Calibration Summary 20151007Wed.txt](https://github.com/correll/advancedrobotics/files/9821/Calibration.Summary.20151007Wed.txt):
###### tenperc: [[0.000384218,0.021906,0.0420431,0.0639975,0.0873286,0.112833,0.142267,0.179014,0.226418,0.292462]]

Therefore I believe the calibration data we got today is acceptable.

-
### Model
[image]

[image]

-
### Image & Blurred Areas
Displaying the blurred areas between two sensors with hand:
[image]

[image]

[image]

[image]

Even in the [Calibrating the Omni 60] (http://occamvisiongroup.com/knowledgebase/calibrating-omni-60/) (Step 7), I still could recognize some blurred areas (red arrows):
[image]

-
### Changing Radius
When I change the Radius, the two most obvious blurred areas could be slightly different ( see where the red arrow points at)

[image]

[image]

-
### Summary
Further areas in the environments like walls, desks look fine but when objects move closer to the Omni 60 blurred areas would become obvious. The ten-percentile reprojection errors we got are less than 0.5 pixels, therefore the results are acceptable according to the instructions on [Calibrating the Omni 60] (http://occamvisiongroup.com/knowledgebase/calibrating-omni-60/).

Thank @jilu7883 for help!

closed 2015-10-08 by BryanBo-Cao

BryanBo-Cao, 2015-10-08:
BTW, the data of the previous comment has already been stored into the camera’s non-volatile EEPROM memory.