correll/advancedrobotics #41: Error when running roslaunch for the stewart platform

rcr2f, 2015-10-07:
Every now and then an error about permissions on the .gvfs file prevents running roslaunch. It seems like roscore or another ros command is corrupting this file. The only fix so far is to either use chown or if that fails, logging out and logging back in works every time.

rcr2f, 2015-11-04:
```
[rosstack] Warning: error while crawling /home/user: boost::filesystem::status: Permission denied: "/home/user/.gvfs"
```
When we check ownership.

```
    d?????????  ? ?        ?            ?            ? .gvfs
```

ikemole, 2015-11-04:
I have seen that happen a couple of times too. Usually I just delete the
".gvfs" and everything works fine again.

rcr2f, 2015-11-04:
I've always logged in and out to fix it. I found the reason why I was getting this issue, every time I'd edit .bashrc, I'd use sudo. This may be a similar reason to why Andy got it too, for using sudo where you don't need to in setting up ROS.

closed 2015-11-04 by rcr2f