correll/advancedrobotics #44: Find the differences between robot_localization package and robot_pose_ekf

rmaheshkumarblr, 2015-10-09:
Reference: http://answers.ros.org/question/191962/state-estimation-and-localization-in-robot-navigation/

robot_localization -> 3D tacking based on Extended Kalman filter with mulitple sensors (multiple odom and multiple IMU..)and provide more functionality  

<b>My comments:  We can have multiple sensors of the same type but there is no visual odometry and camera integration</b>

robot_pose_ekf -> 3D tacking based on Extended Kalman filter fusing wheel odometry, IMU sensor and visual odometry. For tracking pose.

amcl -> 2D localization based on particle filter uses odometry and laser. 

Checking further

rmaheshkumarblr, 2015-10-13:
Robot localization seems to be the better option because it can support multiple sensors including the visual odometry.

"If you are fusing continuous position data such as wheel encoder odometry, visual odometry, or IMU data,set "world_frame" to your odom_frame value. This is the default behavior for robot_localization's state estimation nodes."

Planning to use one of the ros modules for visual odometry.
http://wiki.ros.org/fovis_ros or http://wiki.ros.org/viso2_ros

keerthanpg, 2017-12-04:
1. robot_localization can accept any number of odometry/imu sensor values, one can stack them. However robot_pose_ekf can only take two odometry and one imu data
2. robot_localization lets you fuse sensor data by choosing which variables to consider for current state by mere true/false binaries. However in robot_pose_ekf, this is done by setting the corresponding covariance value to very high, if you want to ignore that sensor or that particular variable.
3. In robot_localization, we can customise the process_noise_covariance and initial_estimate_covariance matrices by setting them to real values allowing greater degree of flexibility and customisation. However in robot_pose_ekf, we can only set them to binary true-false values. In the former, the matrices are 14*14 while in the latter, the matrix is 14*1, further enhancing flexibility in the former compared to the latter

keerthanpg, 2017-12-04:
In robot_localization, we can customise the process_noise_covariance and initial_estimate_covariance matrices by setting them to real values allowing greater degree of flexibility and customisation. However in robot_pose_ekf, we can only set them to binary true-false values. In the former, the matrices are 14*14 while in the latter, the matrix is 14*1, further enhancing flexibility in the former compared to the latter

kkraman02, 2021-08-09:
First of all thanks to both of you for these detailed explanations, I understood the concept and the difference between different approaches very well. Which is very helpful for my research.