correll/advancedrobotics #45: Install Laser Scanner and Configure with ROS Navigation Stack

RohitRaje9362, 2015-10-16:


BryanBo-Cao, 2015-10-16:
Here's how the Laser Scanner from Pepperl+Fuchs looks like:

[image]

[image]

Basic Stats:
Connects through Ethernet Cable, Requires separate External Power (10-30V), has multiple communication configurations using TCP/IP, DHCP.

We were able to ping to it after setting its IP to 0.0.0.0
Still need to read the raw data from ethernet port and then integrate with ROS Navigation Stack.

Thanks @RohitRaje9362

ikemole, 2015-10-22:
The Laser Scanner has been connected, given an IP address and the ROS driver has been setup to publish messages of type **sensor_msgs/LaserScan**. See the wiki for more info.