correll/advancedrobotics #49: Configure Costmap (local_costmap) & (global_costmap)

BryanBo-Cao, 2015-10-16:
[RobotSetup 3.3 Costmap Configuration (local_costmap) & (global_costmap)](http://wiki.ros.org/navigation/Tutorials/RobotSetup)

[costmap_2d](http://wiki.ros.org/costmap_2d)

Sub-issue of #36

Branden-Romero, 2015-12-08:
[image]

Map of ECCR 1B15.

closed 2015-12-08 by Branden-Romero

correll, 2015-12-08:
Does this also work outside?

Branden-Romero, 2015-12-09:
Will try when we get a new usb to serial breakout, can't drive now.

correll, 2015-12-09:
We should have a couple in the lab still

Branden-Romero, 2015-12-09:
By any chance do you know where?

Jorge-C, 2015-12-09:
We already talked to Andy about it and could not find any. Is there any
place we should double check?

correll, 2015-12-09:
John might know

Branden-Romero, 2015-12-10:
[image]
[image]
Got better results for outdoor map. Scan match doesn't find a match consistently so there is ghosting on the trees on bottom right and left.