correll/advancedrobotics #51: Cannot run RViz on the Tegra

ikemole, 2015-10-22:
Running RViz on the Tegra usually throws a **Segmentation Fault** error.

correll, 2015-10-23:
Please follow the instructions in section 3.11 on this page

http://wiki.ros.org/indigo/Installation/UbuntuARM

BryanBo-Cao, 2015-10-23:
Follow the instructions in section 3.11 on http://wiki.ros.org/indigo/Installation/UbuntuARM
```
ubuntu@tegra-ubuntu:~$ rviz
Segmentation fault
ubuntu@tegra-ubuntu:~$ unset GTK_IM_MODULE
ubuntu@tegra-ubuntu:~$ rviz
[ INFO] [1445630742.152709714]: rviz version 1.11.7
[ INFO] [1445630742.153090469]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1445630743.360215923]: Stereo is NOT SUPPORTED
[ INFO] [1445630743.361907611]: OpenGl version: 4.4 (GLSL 4.4).
Segmentation fault
```

Jorge-C, 2015-10-23:
@BryanBo-Cao  and I made some progress, and we were able to trace it back to [these](https://github.com/ros-visualization/rviz/issues/891) two [issues](https://github.com/ros/robot_model/issues/110).

They trace it back to an issue in libpcre (a regexp lib), so we compiled v8.36 from source and got farther: rviz started, but complained about many plugins:
```
[ERROR] [1445639621.540563310]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Failed to load library /opt/ros/indigo/lib/libdefault_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/indigo/lib/liburdf.so: undefined symbol: _ZN7pcrecpp2RE4InitEPKcPKNS_10RE_OptionsE)
```
I'm not sure if those errors are fatal, and they can probably be solved if we compile everything against the newer libpcre. Is it important to get rviz working, especially if we move on to stop using the tegra as a desktop?

Jorge-C, 2015-10-23:
Also, I built it as a `.deb` package, so that it's easy to install/remove:
```
#  sudo dpkg -i ~/pcre-8.36/pcre_8.36-1_armhf.deb
# apt-get remove pcre
```

Jorge-C, 2015-11-25:
Let's close as won't fix. We're moving away from having any GUI run off it.

closed 2015-11-25 by Jorge-C