correll/advancedrobotics #53: Manage development on jetson

Jorge-C, 2015-10-23:
There isn't a lot of space left on the jetson board, we should clean it up and maybe at the same point we can stop using it as a desktop computer (eg having graphics, firefox, code editors...).

Some stuff that uses a lot of space:
- [ ] opencv2 (2 GiB)
- [ ] opencv3 (1.4 GiB) (why do we have this?)
- [x] apt-get cache (1 GiB)
- [ ] Download folder (1.0 GiB)
- [ ] cuda examples (0.5 GiB)

ikemole, 2015-11-20:
I tried to organize the code on the Tegra to be easier to find packages. First, we now have a top-level folder called `~/workspaces/`. Inside this, I created different workspaces (loosely) organized by function. Each *workspace* then contains *packages* related to that functionality. This way, you only build code in the realm that you are developing. Here is what it looks like: 

**~/workspaces/**:

[image]

### Here is what each workspace looks like:

**jaguar_ws**:
[image]

**sensors_ws**:
[image]

**camera_ws**:
[image]

**navigation_ws**:
[image]

**stewart_ws**:
*The code is not yet on the Tegra*

##### The remaining folders are not "workspaces" but are in the same folder: 

**website**:
[image]

**launchfiles**:
[image]

The current size of the `~/workspaces/` folder is **103MB**.

Also, the `~/catkin_ws` has been deleted and all of the new workspaces have been "sourced" in the `~/.bashrc` file. They have been compiled in the new environment and are all functioning as usual.

### Future Goal
We can check in `~/workspaces/` to this Git repository so that people can develop without having to SSH into the robot first. Each person can develop from his computer. When testing is required, the executable will be uploaded to the robot. When it is finalized, both the code and executable can be checked in for deployment. The logistics of this will be worked out. Basically, we do not actually need the source code for the packages on the robot. The only possible problem with this is that, depending on the type of ROS node, it may need to be compiled on the Tegra directly. 

Comments?

ikemole, 2015-11-20:
I also deleted many files and folders that I could verify were no longer relevant. This included stuff in the Downloads folder and some ROS tutorial stuff. I will verify what version of OpenCV is being referenced and delete the other. We can also aim to completely avoid using it as a desktop computer by removing all graphics and stuff like that. Here is what our memory looks like now: 

[image]

I think there is still a lot of stuff to remove and perhaps software to uninstall from the system. 

Here is what the Home folder now contains: 

[image]

**A few questions about folders I didn't understand:**
1. Which folder contains the data for the Wifi Host. Is it `hostap` or `host_wifi_related`?
2. Do we still need `rviz` and `rtabmap` on the Tegra? (Also, why 2 `rtabmap`'s?)
3. What is `pcre-8.36` being used for? Is it still relevant?
4. What is `jaguar_2dnav`? If relevant, can it be placed in a workspace like a ROS node?

Jorge-C, 2015-11-25:
@ikemole pcre is no longer relevant, so I deleted it. Re. your other questions, since nobody has raised any concerns (in particular about hostap, host_wifi_related, and jaguar_2dnav), I think you can go ahead and delete all that too. Did you already delete opencv3?

Another thing I've noticed is that we have a bunch of ros-jade packages. Anybody specifically needs them or can they be removed? Note that we're developing everything in ros-indigo.