correll/advancedrobotics #55: Hardware development for Jaguar

RohitRaje9362, 2015-10-24:
@Zolkin1(Zach) just noticed that the 80-20's in the Correll lab are slightly more larger than the ones we already cut for the Jaguar. 

Images of the two are as below:

[image]
[image]

I think there is not much difference. We have started cutting the larger 80-20's as those are the only ones that closely resemble the ones already cut for the Jaguar.

RohitRaje9362, 2015-10-24:
Also, is there a power-cutting tool in the Correll lab to cut the 80-20's?

correll, 2015-10-24:
Just hacksaw

RohitRaje9362, 2015-10-28:
We are out of angle brackets and appropriate screws. @ryanleonard , @BryanBo-Cao  and I are going to buy the following parts now at McGuckin's.
[image]

To @Zolkin1: Please could you specify if any other parts need to be bought.

BryanBo-Cao, 2015-11-11:
Have these 80-20 angle brackets and appropriate screws been ordered? Currently are we waiting for them for hardware development?

Jorge-C, 2015-11-11:
Andy is working on it. He has ordered hardware and will work on it tomorrow I think.

correll, 2015-11-11:
The hardware is sitting in the CS front office in the 7th floor

Jorge-C, 2015-11-11:
It's already in the lab, between our desks.

BryanBo-Cao, 2015-11-12:
Andy, @Jorge-C and I worked on the 80/20 hardware development based on their design:

[image]

closed 2015-11-15 by Jorge-C