correll/advancedrobotics #56: Embedded Battery Management System for the Jaguar

ikemole, 2015-10-28:
Here are some that I found: 

1. [**Battery Protection Circuit Board for 22.2V 6S Li-ion and Li-polymer battery pack**](http://bestechpower.com/222v6spcmbmspcbforli-ionli-polymerbatterypack/PCB-D132.html) :
[image]

2. [**Protection Circuit Module (PCB) for 6 Cells Li-Ion/Li-Polymer Battery Packs w/ 6A Working & 10A Cut-off**](http://www.all-battery.com/PCBfor6CellsLi-Ion_Li-PolymerPacks_6AWorkingand10ACut-off-32116.aspx?utm_source=GoogleShopping&utm_medium=GDF&gdftrk=gdfV26767_a_7c354_a_7c922_a_7c32116&gclid=CjwKEAjwwbyxBRCS74T049iEp0wSJACkO5v1_YUm29X7rf6ZF3DoGIgB2yxvXdJC7aMnYHMUiDb6MBoCs7Tw_wcB)
[image]

Steve, is this the kind of stuff we need?

ikemole, 2015-10-28:
Also, we could just extend the wires outside the robot using extension wires like [this](http://www.batteryspace.com/connectoradaptorapairoftamiyaconnectorsmalefemale.aspx) or [this](http://www.ebay.com/itm/10-x-20cm-Big-TAMIYA-14AWG-Silicone-wire-Extension-Cable/281577328550?_trksid=p2047675.c100005.m1851&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D1%26asc%3D33430%26meid%3D55258def1f004b1c9074b06015525391%26pid%3D100005%26rk%3D2%26rkt%3D6%26mehot%3Dfs%26sd%3D221672317235). That way, it's easier to charge with our existing equipment without having to remove the battery. Like this:

[image]

mcguire-steve, 2015-10-28:
The BMS is part of the solution. A BMS provides for overcurrent and overvolt protection, but doesn't function as a balance charger. In re the extension idea, that can work, but we'd need to have both Tamiya plugs (the two conductor ones) and the balance JST plugs (the 7 conductor ones) extended to provide a complete harness per battery pack half. 

Maybe something like this could be usable:
http://www.ebay.com/itm/like/252134945802?ul_noapp=true&chn=ps&lpid=82
Small enough and dumb enough to just embed into the system for each battery, with a diode isolation unit on the mains that can handle high amps to the motor controllers.

mcguire-steve, 2015-10-28:
The external power plug would just then be something like a high-amp DC input at say 15V: 
The charger draws 80W, so two of them draw 160W - maybe something like: http://www.trcelectronics.com/View/Mean-Well/LRS-200-15.shtml
to have some margin. The pins used to transfer external power would need to handle the combined amperage of 13.3 A, so a minimum 15A rating if only two pins are used, 7.5 if we split it up across four pins (2 supply, 2 return).

mcguire-steve, 2015-10-28:
Might also be useful to open the cases of the chargers and reroute the status LEDs to the outside for monitoring purposes.