correll/advancedrobotics #57: Run motor driver and IMU simultaneously

BryanBo-Cao, 2015-10-28:
Motor driver and IMU are connect to Jetson board via the same FTDI. We attempted to run motor driver and receive IMU data simultaneously but failed. The baud rate of motor drive is 115200 baud while IMU's baud rate is 57600 baud.
[image]

---
One of the solutions could be to change motor driver's or IMU's baud rate to the same rate, such that, however, I suspect we might need to add a module for management of these data since they use the same baud rate via the same FTDI.

The other simple solution would be just to buy another FTDI: https://www.sparkfun.com/products/9873, so that motor driver and IMU can be connected to Jetson separately. Branden and I would suggest this solution.

Thank @Branden-Romero & @RohitRaje9362 for help!

Anyone has better ideas?

correll, 2015-10-29:
We have a ton of FTDI cables in the lab and I gave already some to somebody
a while back.

rmaheshkumarblr, 2015-10-29:
Why are we using the same FTDI?  2 weeks back I set a separate FTDI for the
motor and the IMU.. Did we change something?

rmaheshkumarblr, 2015-10-29:
Re-attached the FTDI we were using earlier and it works fine now.

closed 2015-10-29 by rmaheshkumarblr