correll/advancedrobotics #59: Mount Jetson on Jaguar

BryanBo-Cao, 2015-10-29:
[image]

[image]

[image]

Thank @RohitRaje9362 for help!

Now Jetson could be powered on by the battery again.

@Branden-Romero I think now you could move it around by joystick to measure odometry without worrying about the FTDI cable drawning into the gears...

closed 2015-10-29 by BryanBo-Cao

BryanBo-Cao, 2015-11-04:
Share the videos from last week:
[Robot moving outside the lab VID 20151029 142200 ](https://youtu.be/bMI5y4v7eDU)
[Robot moving on stones and grass with lego figure VID 20151029 170844 ](https://youtu.be/d7ZqrxK_nzg)
[Robot moving with a human and lego figures VID 20151029 173740](https://youtu.be/lBaOTfgMe3Q)

RohitRaje9362, 2015-11-05:
Video of Jaguar climbing stairs successfully:
https://www.youtube.com/watch?v=ajecEemVHm4