correll/advancedrobotics #61: Stewart platform: control

Jorge-C, 2015-10-31:
The platform doesn't track pose commands very well (it has too much stationary error, probably because some measurements aren't that great), so we need to stabilize it.

Given that we have an inverse model, we can use feedforward control, but that still has stationary error -> we need to add feedback. This is the type of architecture: http://computationalnonlinear.asmedigitalcollection.asme.org/data/journals/jcnddm/927599/m_cnd_009_01_011014_f001.png

Anybody with interest or experience in controls, feel free to pitch in with suggestions :)

correll, 2015-10-31:
Is this about the pose after taking IMU inputs into account? The servos should be pretty precise.

Jorge-C, 2015-11-03:
Some extra PID tuning would be nice to have, but this is working now.

closed 2015-11-03 by Jorge-C

correll, 2015-11-03:
This is a simple method to get good pid values

https://en.m.wikipedia.org/wiki/Ziegler–Nichols_method