correll/advancedrobotics #63: Internet connectivity to display location in Google Maps

rmaheshkumarblr, 2015-11-03:
I am able to use ros slave node to communicate with the ros master node to get the latitude and longitude from the GPS module. Basically the ros slave node on my PC is able to get the latitude and longitude using the roslibjs. 

I am just wondering how I will be able to display the location on the map because the Google API needs internet connectivity and I won't have access to Internet on my laptop when I am connected to the Jaguar Access Point. Is it possible to get it working without having another USB wireless module attached to my laptop. Anyone has any suggestions?

correll, 2015-11-03:
An easy fix would be to use wired ethernet on your laptop to.

BryanBo-Cao, 2015-11-03:
Have you tried Virtual Machine to access the Internet? Use you host machine, let's say OS X, to connect to Jaguar while opening a Virtual Machine running another OS, say Windows, as a guest machine to modify the network configuration to access the Internet. At this point, the Google Maps APIs might be used on this guest machine.
Reference: http://www.virtualbox.org/manual/ch06.html#network_internal

I haven't tried this before but I was just thinking if this is possible :)

rmaheshkumarblr, 2015-11-03:
Used the ethernet cable for internet connectivity.

The Virtual Machine part does not help to provide internet connectivity because there is only one network device and hence it is able to connect only to one network at a time.

closed 2015-11-03 by rmaheshkumarblr