correll/advancedrobotics #64: Displaying the GPS Latitude and Longitude on a browser / google maps.

rmaheshkumarblr, 2015-11-03:


rmaheshkumarblr, 2015-11-03:
[image]

Process to get it working on your laptop:

1) Connect to jaguar_ap 
2) SSH to 192.168.205.1 ( IP address of Jaguar )
3) Start roscore
4) Start the GPS launch file or all the services including the GPS
5) Start the rosbridge service ```roslaunch rosbridge_server rosbridge_websocket.launch ```
6) Make sure you have internet connectivity on your laptop
7) export ROS_IP="IP you were assigned by the jaguar AP"
8) export ROS_MASTER_URI=http://192.168.205.1:11311
9) Open the html file that is uploaded to the code repository.

correll, 2015-11-03:
Great!

ikemole, 2015-11-04:
@rmaheshkumarblr I've incorporated this into a single page with the camera feed.

closed 2015-11-07 by rmaheshkumarblr