correll/advancedrobotics #65: Create jaguar webpage

ikemole, 2015-11-04:
The goal is to create a single page for the jaguar where we can see:
- GPS Location on a map
- Camera Feeds

To display the camera feeds, we use a ROS package called [web_video_server](http://wiki.ros.org/web_video_server). This launches a web server on the tegra that will subscribe to any image topic and display the published image on a webpage.

correll, 2015-11-04:
I'd recommend having the we server on the client that connects to tegra via ROS (same ros core).

This would allow to free up some resources and reduce bandwidth as the camera feed can be compressed (via ROS).

ikemole, 2015-11-04:
Actually we would not be installing a full-blown web-server. We would just be using the simple HTTP server that comes with python to serve static files from a folder. Also, it would be a html file and some Javascript. So, inside the folder, we just type `python -m SimpleHTTPServer` and we can view the HTML file on port 8000 of the Tegra. It's very lightweight.

ikemole, 2015-11-04:
The page has been created. Here is what it looks like:

[image]

The webpage's code is located in `~/catkin_ws/src/jaguar_website`. 

To start it, navigate to the folder and run the `launch_website.sh` script. This script is for convenience as it just launches the Python SimpleHTTPServer and makes the page available on the `jaguar_ap` network. 

In a web browser on your computer, navigate to `192.168.205.1:8000` to view the page.

Note that the following packages need to be started before viewing the page:
- rosbridge
- web_video_server
- gps

correll, 2015-11-04:
Cool!

closed 2015-11-04 by ikemole