correll/advancedrobotics #68: Jaguar battery cells unbalanced

Branden-Romero, 2015-11-10:
[image]

Jorge-C, 2015-11-10:
Does running cell balancing not fix it?

Branden-Romero, 2015-11-10:
It fixes it, but this is the second time after use that the battery has become unbalanced since we attached the laser scanner.

Jorge-C, 2015-11-10:
We should recharge more often. A lot of low battery alarms for RC
airplanes/quadcopters go off at about 3.3V per cell (under load, so 3.2V
with no load is much more problematic).

Re. unbalancing by discharge, that might simply suggest that the BMS of the
jaguar is not great. Does it even have a Battery Management System?

ikemole, 2015-11-10:
Just an observation...

I've noticed that the same cell seems to discharge faster than the other
five cells. Every time I was about to charge it, the lowest number always
shows on the bottom left position on LCD (as in this picture). I am
guessing that the same cell is represented in that location on the screen.

I am not sure how relevant that is.

mcguire-steve, 2015-11-10:
The Jaguar does not have a battery management system to prevent
over-discharge or over-charging. Cell unbalancing is related to
manufacturing differences between individual cells - the fact that a single
cell is perpetually the weak link is consistent.
Long-term solution would involve a ROS node with some A/D hardware to
actively monitor battery voltage and current draw so that we can estimate
the amount of charge remaining and not damage the LiPo packs due to
over-discharge.
-Steve

Jorge-C, 2015-11-25:
Closing as it is just an artificat of not having a BMS ( #56 ).

closed 2015-11-25 by Jorge-C