correll/advancedrobotics #71: Integrate all hardware components

BryanBo-Cao, 2015-11-23:


BryanBo-Cao, 2015-11-23:
[image]

[image]

[image]

However, we still need to unmount stewart platfom to charge the battery unitl #56 (Embedded Battery Management System for the Jaguar) is solved.

So far I have focused on hardware for weeks, I will concentrate more on web-map in the future.

closed 2015-11-23 by BryanBo-Cao

correll, 2015-11-24:
This beast looks awesome!