correll/advancedrobotics #72: Turn on Jetson automatically when turning on Jaguar from battery

BryanBo-Cao, 2015-11-23:
Currently we have to turn on the power by the button on the back of Jaguar, and then press the power button of Jetson inside Jaguar, which is extremely inconvenient.

RohitRaje9362, 2015-11-23:
I suppose the Jetson will turn on automatically as long as it's reset
capacitors stay charged i.e. as long as the battery is always connected and
the Jetson is not explicitly shut down.

However, after removing the batteries for charging, restarting the Jetson
does become an inconvenience.

Rohit

mcguire-steve, 2015-11-23:
The small button on the front panel next to the main power switch is wired
to a header for just such a reason. The LED inside and the button connect
to the 'front panel' header block on the Jetson.
-Steve

correll, 2015-11-24:
Maybe it's possible to bridge the power button?

Jorge-C, 2015-11-25:
The small button works.

closed 2015-11-25 by Jorge-C