correll/advancedrobotics #73: Quick stewart platform update

Jorge-C, 2015-11-25:
A few issues have cropped up when integrating systems:

- [ ] Edit razor nodes so that they boot up with different names, can't have both razors publishing together (even in different channels).
- [ ] I couldn't read any data from the jaguar imu (its serial port was quiet), not even when it was running alone (?)
- [ ] Platform needs to get 12V from the jaguar (I was testing with it plugged in to a wall)

BTW, please charge the batteries before using it in the next few days, I charged it for a while but it still needs a lot of juice to be at 100%.

Also, Steve configured functional internet on the jetson, which makes it much more convenient to work with it.

rcr2f, 2015-12-01:
For the razor node names, could we just change the name in imu_node when it's initialized? Or is it getting the names from firmware?

rmaheshkumarblr, 2015-12-02:
The node names can be changed to the name we want using the XML syntax in
the ROS launch file.

http://wiki.ros.org/roslaunch/XML/remap