correll/advancedrobotics #76: Video Creation

rmaheshkumarblr, 2015-12-07:
Our first effort in video creation. Thanks @ryanleonard for taking the video.

We can edit/create new one if necessary based on suggestions.
Right now I have not edited the audio :P 

https://youtu.be/1g8PVeA4QE4

BryanBo-Cao, 2015-12-08:
Just curious what name would you like for the Jaguar and Stewart Platform? I think it would be better to display like "Introducing [Delivery Robot]" near the beginning of the video.

BryanBo-Cao, 2015-12-10:
Hey guys just edited the the video https://youtu.be/Tb0sU9bNYIY.

In my opinion, the video could be better if more video scripts could be filmed (some of them may already be filmed before but it's better to film again):
1. Someone (like Rebecca?) tag the phone to order a coffee;
2. Rebecca receives the coffee;
3. Film the screen showing Jaguar creating the map of its surrounding when moving;
4. Climb stairs;
5. Avoid humans.

@ryanleonard it would be appreciated to get the original film files from your camera.

correll, 2015-12-10:
Cool! In my opinion we should focus on outdoor scenes and order the cafe to
the math building. Outdoor and stairs are all novelties that will get a lot
of interest. Indoor footage not so much.

BryanBo-Cao, 2015-12-10:
Some of the films of the exterior of the Jaguar in this version should be replaced with those.

correll, 2015-12-10:
You guys are actually on a very steep trajectory! The robot looks great and
I have seen a lot of technical progress toward autonomy.

I'd add to the list:

- replace "self-turning" with "autonomous navigation"
- show the map every now and then, maybe even with a path planned therein
- the horizontally stable coffee should only be mentioned if this actually
works. Maybe the robot could go up a not too steep ramp if it does not work
on stairs. We can also add a ring around the top of the Stewart Platform
- it would be awesome if we get LSD SLAM to work. In the worst case things
should be stored in a bag file and we do processing offline. Showing that
there is a 3D map would be cool.
- show the position on a google map when outside

Xavier is coming tomorrow and will bring a color version of his camera. He
will also be able to help with calibration.

N

@ryanleonard <https://github.com/ryanleonard> it would be appreciated to

> get the original film files from your camera.